heethesh/lidar_camera_calibration #5: Could this be used with real time data?

dejongyeong, 2020-06-12:
Hi, I would like to know does this package work if I doesn't use rosbag? I'll launch camera and velodyne in different terminal that publish the require topic as mentioned in the README.md.

Thanks.

heethesh, 2020-06-12:
Yes it should work, you can still launch the GUI and pick (and save) corresponding 2D/3D points.

dejongyeong, 2020-06-12:
Hi @heethesh, thanks for getting back to me, do I need to save the calibrated YAML file?

heethesh, 2020-06-12:
Yes, if you save it in `/home/.ros` folder (not sure) - which is the default save path, I believe ROS will automatically pick the data from here for a given camera name.

dejongyeong, 2020-06-12:
Ok @heethesh Thanks for the information. I'll give it a try and see how it goes. I might need some pointers if I have some difficulties. Thanks.

heethesh, 2020-06-12:
No problem, you may want to update the point-cloud visible ranges for the GUI [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L232). Working on a new update that uses Rviz GUI instead of Matplotlib which should be more convenient to work with.

dejongyeong, 2020-06-12:
Ok. I'm new to ROS and will look into details once I got that to work. Thanks.

dejongyeong, 2020-06-12:
Hi @heethesh, one quick question that I would like to ask, does it work for LiDAR 32 channel?

heethesh, 2020-06-12:
yes it should work