heethesh/lidar_camera_calibration #6: solve error

crankler, 2020-06-13:
Hi, when I have choosen the corner points in image and lidar data, the error shown as follow:
> ...
  File "/home/../catkin_test/src/lidar_camera_calibration/scripts/calibrate_camera_lidar.py", line 320, in calibrate
    assert(points2D.shape[0] == points3D.shape[0])
AssertionError

Then I print the data:
> ...
points2D.shape[0]:  32
points3D.shape[0]:  43

Can you tell me how to rectity this problem? thank yuo!

closed 2020-06-13 by crankler

heethesh, 2020-06-25:
Deleting the .npy files where the point are stored will reset the counts

kubs0ne, 2021-04-12:
Hi, I had the same error, and deleted the files as you said, but now I have a different error: 

> IOError: [Errno 2] No such file or directory: '/home/kuba/catkin_ws/src/lidar_camera_calibration/calibration_data/lidar_camera_calibration/img_corners.npy'

 Which indicates that the initial npy files are needed. Do you have any further suggestions of what could I do? 

Regards

heethesh, 2021-04-12:
did you delete the file when the calibration program was still running?

kubs0ne, 2021-04-13:
No, actually I didn't do it so it is strange. I downloaded the repo again and I have the same 2 errors, however I select 5 points, just like you in the youtube film.