heethesh/lidar_camera_calibration #9: Using .PCD files

Barath19, 2020-07-22:
Hi,
is it possible to calibrate with .pcd file as an input?!...could you provide any useful theory / blogs for lidar calibration.

heethesh, 2020-07-23:
The `calibrate` function itself just takes in corresponding 2D/3D points to estimate the pose. If you want to use the picker GUI, just get rid of all the ROS stuff and use the code from where the point cloud is in Numpy format. You may use Open3D to load your `.pcd` file and convert it to `np.ndarray`. Note, you may need multiple poses (of a checkerboard, possibly multiple pcd files) or multiple correspondences in the same point cloud/image should also work!

The theory is based on [PnP algorithms](https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html) and optionally non-linear refinement using LM optimization (see [Factor Graphs](http://www.cs.cmu.edu/~kaess/pub/Dellaert17fnt.pdf) book Ch 2/6).

closed 2020-07-23 by heethesh

Chri1996, 2022-12-08:
Is there a repo, where someone has changed the code from ros to pcd/img pairs?