heethesh/lidar_camera_calibration #10: Error Metrics

gcarlin8, 2020-08-03:
Hello heethesh,

Is there anyway to get the information about the point error/pixel error or any other error metrics

heethesh, 2020-08-12:
Since you manually pick 3D/2D correspondences here, you can re-project the picked 3D points to the image after you have the extrinsics estimate and compute the pixel error with the 2D points you initially picked. Also have a look at OpenCV's PnP methods [documentation](https://docs.opencv.org/master/d9/d0c/group__calib3d.html#ga50620f0e26e02caa2e9adc07b5fbf24e) which lets you set the inlier count, reprojection error thresholds, confidence, and other optimization criteria.

heethesh, 2020-10-26:
See my comment [here](https://github.com/heethesh/lidar_camera_calibration/issues/22#issuecomment-715589902)