heethesh/lidar_camera_calibration #12: Does not work when using 64-channel LiDAR

JihunDev, 2020-09-11:
## Description of the situation
I am using Ouster OS1 64-Channel Lidar.
I am trying to calibration to my USB camera and Lidar. However, it outputs as below

[image]

## Question
I have 2 questions.
1. Does this code work on Velodyne devices?
2. Does it work only on 16 channels?

## Development environment
- ubuntu 18.04 LTS
- Python 2.7
- ROS
- MatplotChart 2.1.2

## Resources
[Ouster OS-1](https://ouster.com/products/os1-lidar-sensor/)

heethesh, 2020-09-11:
Number of beams/sensors should not matter as long as the message format is `PointCloud2`. Just remove `c=cmap(colors)` on line 256 and try running it again. Also can you check the size of `points` wtihin `extract_points_3D`?

saran1995, 2020-10-23:
Hey @JihunDev  my situtaiton is similar to yours:
1) We are using  Ouster OS1 64-Channel Lidar and we are trying to calibrate usb camera and Lidar. 

Once we run the command for calibration the gui doesn't not opens up, basically its stuck on rospy.spin
[image]

**Development Environment** 
- ubuntu 18.04 LTS
- Python 2.7
- ROS
- MatplotChart 2.1.2

Please guide how did you solved this issue.

heethesh, 2020-10-23:
@saran1995 also see #16

closed 2020-10-26 by JihunDev