heethesh/lidar_camera_calibration #13: Which target to use the calibrate the camera?

gonultasbu, 2020-09-20:
Hello,

I recorded a rosbag while showing a regular checkerboard template with 25mm squares a a calibration target to the camera, however the camera calibration script doesn't seem to do anything with that rosbag. It just plays the rosbag, finished it, and restarts. Corresponding calibration data folders are empty.

Thank you.

heethesh, 2020-09-20:
You can use any target supported by the ROS [camera_calibration](http://wiki.ros.org/camera_calibration) package. Update the parameters you use in this [line](https://github.com/heethesh/lidar_camera_calibration/blob/master/launch/camera_calibration.launch#L5) and run the following commands. The camera calibrator GUI should show up. The calibrated parameters are stored at `~/.ros/camera_info`.

```
sudo apt install ros-melodic-camera-calibration # or ros-kinetic
roslaunch lidar_camera_calibration camera_calibration.launch
```

gonultasbu, 2020-09-20:
Turns out I needed to adjust the parameters (square size and checkerboard size) indeed and hold the calibration target closer. Thanks for the clarification!

closed 2020-09-20 by gonultasbu