heethesh/lidar_camera_calibration #15: Few questions related to lidar camera calibration

poornimajd, 2020-09-29:
Hello 
For the lidar camera calibration , I created a bag file by using a launch file to run both velodyne lidar and zed camera together and hence record the topics  /velodyne_points ,/camera_info and /left_image_rect_color .Using the calibrate_camera_lidar.py , I tried to get the 2d and 3d correspondences,but when I run the display_camera_lidar_calibration.launch, it displays the calibration but I get a error that none of the frames of the two sensors were synchronized till now.
In regard to this I have the following questions
1)While recording the rosbag file , am I suppose to move the checkerboard to different positions like how we do in camera calibration , or should I just record one pose,such that the checkerboard appears in both the camera and lidar frame
2)The way I have described the recording of the rosbag file (earlier in the first paragraph) , is that the way to record it?  Then why do I get that error , or do you have any suggestion as to how to deal with it?
3)I got the following windows of 3d and 2d checkerboard(shown below), the 3d points are not actually clearly visible,is there any reason behind it?
[image]
I just commented the following line ,because I was getting an error .(Axes3dSubplot has no attribute set_facecolor)
https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L253

Sorry for the lengthy question,any suggestion is greatly appreciated.
Thanks in advance

heethesh, 2020-09-29:
You should update the point cloud field-of-view to display in this line https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L232 as mentioned in the README. Matplotlib GUI does not handle a large point cloud very well and also makes it harder to use it for picking points, set the suitable FOV that works for you. Similar updates required for displaying the final calibrated data.

heethesh, 2020-09-29:
Regarding the matplotlib error, try upgrading it to >2.0 `pip install -U matplotlib`

heethesh, 2020-09-29:
You should record several poses of the board when it is visible both in the camera and LiDAR view. Regarding the timestamps not matching in your bag file, I'm not sure about that - do they play properly in Rviz when you run `rosbag play`? Try increasing the `slop` value [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L494) which is used in syncing messages. You can also run this application on live sensor data.

poornimajd, 2020-09-29:
Thanks for the quick and detailed reply!
> Regarding the matplotlib error, try upgrading it to >2.0 `pip install -U matplotlib`

I have version 3.3.2,anyway I will just continue with the commented line.Thanks

poornimajd, 2020-09-29:
>  do they play properly in Rviz when you run `rosbag play`?

Yes it is playing properly in rviz.

>  Try increasing the `slop` value [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L494) which is used in syncing messages. You can also run this application on live sensor data.

Yeah I will try with this.
Thanks for the suggestion.

heethesh, 2020-09-29:
Yes, they should be in meters

surajm9, 2022-06-14:
Hello @poornimajd Did you figure out the solution for this issue? I am stuck at the same point.
@heethesh Can you please guide and help me solve this? 

Thank you in advance!

huynhthaihoa, 2022-10-25:
Hello @poornimajd, could you please share how you made the launch file to run both the Lidar and camera together? Thank you so much