heethesh/lidar_camera_calibration #20: Bag file record lagging behind when using roslaunch lidar_camera_calibration play_rosbag.launch bagfile:=calinput.bag

Bsting118, 2020-10-15:
So I have a bag file in my workspace folder named calinput.bag. When I run the command, roslaunch lidar_camera_calibration play_rosbag.launch bagfile:=calinput.bag in my Ubuntu 16.04 terminal, it initially lags behind with time; taking around 7-12 seconds to play a steady speed(like the recording is freezing and then actually plays 10 seconds later; observed when doing  rosrun image_view image_view image=/gmsl_camera/port_0/cam_0/image_raw). This lag with the rosbag recording might also be affecting my final calibration launch file when I want to preview the lidar calibrated with the camera. The lidar lines and camera just don't align or overlap as much as they should.

heethesh, 2020-10-15:
As suggested before, try setting the `--rate` flag in [rosbag play](http://wiki.ros.org/rosbag/Commandline#play) to something lower than 1.0.

Bsting118, 2020-10-15:
Ok, I will try that soon!

Bsting118, 2020-10-19:
For the python solution using the python virtual environment, @heethesh what Python should I use? Python 3 or Python 2?

heethesh, 2020-10-19:
@Bsting118 Please follow-up only on one thread (preferably just this issue here on GitHub for all your queries).
It does not really matter which Python version you use, I would pick Python 3 as Python 2 is no longer supported and might not have the latest OpenCV libraries.

Bsting118, 2020-10-19:
Ok, I will ask questions in this thread and thank you for the input!

Bsting118, 2020-10-19:
@heethesh I made the Python 3 virtual environment successfully in my workspace folder/directory. However, I tried doing the pip install opencv-python and it gave me an ImportError saying "No module named 'skbuild' ". It also gave red text saying "Command 'python setup.py egg_info' failed with error code 1 in /tmp/pip-build-kboewfzm/opencv-python/". Why did it give me these errors? My pip version is 8.1.1.

heethesh, 2020-10-19:
Follow this https://pypi.org/project/opencv-python/

Bsting118, 2020-10-19:
I followed the guide and after doing $ pip install opencv-python it gives me a whole block of red text with the main error reading: "Problem with the CMake Installation, aborting build. CMake executable is cmake; ERROR: Failed building wheel for opencv-python". Seems my CMake is causing issues somehow. My Python version is 3.5 for Python 3 and Python is 2.7; I updated my pip in my virtual env so maybe my Python 3.5 is not supported? How do I fix this without updating my global Python version? Can you update Python ONLY within the virtual environment?

heethesh, 2020-10-19:
Try this instead `pip install opencv-python==4.2.0.34`

Bsting118, 2020-10-19:
Ok I will try that soon, thank you!

Bsting118, 2020-10-20:
I got some Proxy errors and failed-retries after executing those above pip commands; in fact anything I do with pip now does the failed/retry. I am using a PX2 currently so my config is a little unusual; its an arm architecture.

Bsting118, 2020-10-20:
It keeps saying, "WARNING: Retrying (Retrty(...)) after connection broken by 'ProxyError('Cannot connect to proxy.', NewConnetionError'..." Then does that like 4 times and then says "ERROR: Could not find a version that satisfies the requirement opencv-python==4.2.0.34(from versions: none).

Bsting118, 2020-10-20:
I think its because I didn't connect to the internet via ethernet. I will do that on my Nvidia PX2 first and then try your pip commands in my virtual environment again.

Bsting118, 2020-10-20:
It says there is no version that satisfies the requirement for pip install opencv-python==4.2.0.34. The 4.3.0.38 was recognized BUT it gave me a block of red text with the Error reading: "Problem with the CMake installation, aborting build. CMake executable is cmake. ERROR: Failed building wheel for opencv-python." Is there another command I could do? Could it be the directory that my virtual environment is in be causing this cmake error problem?

Bsting118, 2020-10-20:
After doing $ pip install opencv-python==4.3.0.38:

ERROR: Command errored out with exit status 1:
   command: /home/nvidia/ros_l-c_cal_workspace/env/bin/python3 /home/nvidia/ros_l-c_cal_workspace/env/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmp60dd0nvb
       cwd: /tmp/pip-install-vjicr_iu/opencv-python
  Complete output (10 lines):
  /tmp/pip-build-env-dbh61f2w/overlay/lib/python3.5/site-packages/cmake/data/bin/cmake: error while loading shared libraries: libssl.so.1.1: cannot open shared object file: No such file or directory
    File "/tmp/pip-build-env-dbh61f2w/overlay/lib/python3.5/site-packages/skbuild/setuptools_wrap.py", line 560, in setup
      cmkr = cmaker.CMaker(cmake_executable)
    File "/tmp/pip-build-env-dbh61f2w/overlay/lib/python3.5/site-packages/skbuild/cmaker.py", line 95, in __init__
      self.cmake_version = get_cmake_version(self.cmake_executable)
    File "/tmp/pip-build-env-dbh61f2w/overlay/lib/python3.5/site-packages/skbuild/cmaker.py", line 82, in get_cmake_version
      "Problem with the CMake installation, aborting build. CMake executable is %s" % cmake_executable)
  Traceback (most recent call last):
  
  Problem with the CMake installation, aborting build. CMake executable is cmake
  ----------------------------------------
  ERROR: Failed building wheel for opencv-python
Failed to build opencv-python
ERROR: Could not build wheels for opencv-python which use PEP 517 and cannot be installed directly

heethesh, 2020-10-21:
You are probably on a different architecture where opencv-python is no pre-built for. You should check the platforms pre-built [here](https://pypi.org/project/opencv-python/4.3.0.36/#files) in the `Download Files` tab. Also check the release history tab for other version. If none of these work for you, OpenCV python lib needs to be built from source or check the Nvidia website if they release a version for your platform. It is easier if you work on a x86/amd64 platform running Ubuntu 18.