heethesh/lidar_camera_calibration #21: Created New Bag File To Test and Now Won't Stop or Pause to Pick Points At All

Bsting118, 2020-10-21:
Hello Heethesh,

I created a new bag file with the same name as my old bag file(though I deleted my old bag file) and now the lidar and camera calibration doesn't seem to be working. It's not working because now when I pause the bag file after doing play_rosbag.launch and then I press ENTER to pick points after pausing, it won't display the two new windows to allow me to pick points. It seems the program just keeps going without doing the ENTER to pick points. I waited until the end of the rosbag play and it didn't do anything. Is there something else I needed to change when making a new bag file that may be causing this issue? Any launch files, scripts, or python code that I need to update or change to adjust to the new bag file? The new bagfile has the same name as the old one so I am not sure what is going on. Also, does being connected to the internet a requirement when performing lidar and camera calibration(because I am not plugged into the ethernet since I had to use my ethernet port for my velodyne lidar)?

Please get back to me on this issue, thanks!

(same software versions, only python virtual environment added to workspace folder)

heethesh, 2020-10-21:
Can you refer to issue #16 to debug if the topic names are correct and the callback function gets called?

Bsting118, 2020-10-21:
I can only operate on the Drive PX2 or aarch64 due to my research project primarily using it and cannot change it.