heethesh/lidar_camera_calibration #24: Can This Calibration Process Be Used on ROS Melodic

Bsting118, 2020-12-08:
Hello,

I was wondering if this lidar-camera calibration process I got working can be used on ROS Melodic (aka, Ubuntu 18.04)? If it can be used on ROS melodic, is there anything I need to change with the files or setup? Thanks!

Cordially,
Brendan Sting

heethesh, 2020-12-08:
Yes it should work on Melodic. I haven't tested it but I also don't think it has any dependencies on Kinetic.

enting8696, 2021-04-16:
i'm ubuntu melodic  18.04
its work !
just install some package like ros-numpy or tf2.....

rupikaraj, 2023-04-06:
> Yes it should work on Melodic. I haven't tested it but I also don't think it has any dependencies on Kinetic.

I am using your code for OS1 and usb camera on ROS melodic. I picked 5 corresponding points on image and pcd. The values got updated to corresponding .npz files. But when I displayed the fusion, it was not correct. I have followed all your instructions also. 

Should I pick more corresponding points ?

heethesh, 2023-04-07:
@rupikaraj yes, please try picking more points around 20-30

rupikaraj, 2023-04-07:
> @rupikaraj yes, please try picking more points around 20-30

Hi @heethesh  thankyou for replying. I'm still confused tho. Should I pic more coordinates other than the checkerboards corners in the same window or ... Did you mean to do the calibration with corners of checkerboard on multiple locations selecting only the corners on each calibration step ? 

Also.. even tho the calibrated parameters were wrong, I tried the fusion step in real-time. The lidar coordinates do get projected to the image. But the process is taking a lot of time. The fused image frames have some kind of time lag for about (20-30) seconds and I don't understand why this is happening!