heethesh/lidar_camera_calibration #25: The issues of point cloud

yulan0215, 2021-01-03:
Hi!
I used this package and I used Ubuntu 16.04 ROS kinetic and the version of python is python 2.7. The lidar I did not use velodyne, I used ouster data but I changed the topic into /velodyne_points, however, when I did the extrinsic calibration between camera and lidar, the issues occurred and the window of point cloud did not show, only the image window. 
[image]
Thx!

heethesh, 2021-01-04:
Seems like you need to resize the numpy arrays here (stack the first two axes) before displaying, similar issue #12, also see this [comment](https://github.com/heethesh/lidar_camera_calibration/issues/16#issuecomment-702012747).

heethesh, 2021-01-21:
This [solution](https://github.com/heethesh/lidar_camera_calibration/issues/26#issuecomment-764980653) seems to work for Ouster LiDARs

heethesh, 2021-01-21:
Support for Ouster LiDAR has been added now, set `OUSTER_LIDAR = True` [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L75).

closed 2021-01-21 by heethesh

yulan0215, 2021-01-24:
> Support for Ouster LiDAR has been added now, set `OUSTER_LIDAR = True` [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L75).

Thx,this really help