heethesh/lidar_camera_calibration #26: point cloud tab only show a few points

snakehaihai, 2021-01-21:
Hi
thanks for the work
I`ve read through all the posts and solved some of the bugs such as the color issues. mpl versions issue > 2.0

I`m using os 1 ldiar with 10hz rate and camera at 10 hz as well. 
I adjust the slop between 0.1 to 0.5 and  change the inrange function of ldiar to various number. but i can't get the full point cloud to display properly
[image]

I eventually removed all teh inrage and tried to display all points. but problem it is still 2points. May i ask what could be the issue?

snakehaihai, 2021-01-21:
hi 
I found out that ouster lidar has shape of (16, 1024, 9)
is it because of this that we have issue?

heethesh, 2021-01-21:
Yes, it could be. Looks like the Ouster LiDAR data is organized or grouped by the beams. Does your sensor have 16 beams?

snakehaihai, 2021-01-21:
Thanks for the quick response. Yes my lidar is 16 line lidar. I saw your video  has 2 number 384, 4. what does those mean?

heethesh, 2021-01-21:
I believe that was the intensity channel data, which is only used for visualization colors [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L246). Just make sure the `points` array has N rows, corresponding to all points from all beams, and the first 3 columns represent X, Y, Z data. You may just disable the color map if you don't have an intensity channel. Apologies, I have not tested this on Ouster data, this was only tested on a Velodyne point cloud.

snakehaihai, 2021-01-21:
the 9 means  x y z intensity t reflectivity ring noise range
So i should merge 16 ring with each 1024 points into one single 1024x16 points?  
then behind remove t reflective ring noise range?

heethesh, 2021-01-21:
Try this out `points = points.reshape(-1, 9)[:, :4]` add this on line 230

snakehaihai, 2021-01-21:
> Try this out `points = points.reshape(-1, 9)[:, :4]` add this on line 230
this  is the correct solution i can see the depth now

closed 2021-01-21 by snakehaihai

heethesh, 2021-01-21:
Updated master now. If you are using Ouster LiDAR, set `OUSTER_LIDAR = True` [here](https://github.com/heethesh/lidar_camera_calibration/blob/master/scripts/calibrate_camera_lidar.py#L75).