heethesh/lidar_camera_calibration #28: KITTI DATASET(left_camera and VLP64E)

enting8696, 2021-02-02:
Thanks to the author for providing great code ！
I tried to use the kitti dataset (left_camera and VLP64E) .
May I ask why I changed the topic to /kitti/camera_gray_left/image_raw
And run  **roslaunch lidar_camera_calibration display_camera_lidar_calibration.launch** 
[image]
The result of the image does not cover the entire window ? it feels crooked.
[image]
Does this algorithm apply to the left camera?

The display in rviz is normal
[image]

heethesh, 2021-02-02:
Make sure to set the field of view of the point cloud while projecting it. See the README for more details.

enting8696, 2021-02-03:
Thank you for your intentions and replies, thank you, and wish you success in your research

closed 2021-02-03 by enting8696