heethesh/lidar_camera_calibration #32: i got an issue in running roslaunch ros kinetic

john-le1996, 2021-03-26:
roslaunch lidar_camera_calibration play_rosbag.launch bagfile:=lidar_camera.bag

... logging to /home/asd/.ros/log/523261e0-8e4a-11eb-8ccb-04ea56f154b9/roslaunch-asd-2118.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://asd:37717/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    player (rosbag/play)

auto-starting new master
process[master]: started with pid [2128]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 523261e0-8e4a-11eb-8ccb-04ea56f154b9
process[rosout-1]: started with pid [2141]
started core service [/rosout]
process[player-2]: started with pid [2145]
[ INFO] [1616773542.810591800]: Opening /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag
[FATAL] [1616773542.810886481]: Error opening file: /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag
[player-2] process has died [pid 2145, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play --clock --loop /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag __name:=player __log:=/home/asd/.ros/log/523261e0-8e4a-11eb-8ccb-04ea56f154b9/player-2.log].
log file: /home/asd/.ros/log/523261e0-8e4a-11eb-8ccb-04ea56f154b9/player-2*.log

heethesh, 2021-03-26:
check your rosbag file again, can you play with the command `rosbag play FILENAME`?

john-le1996, 2021-03-26:
yes i can play may bag using command rosbag play filename

john-le1996, 2021-03-26:
i build you repo sir 

Base path: /home/asd/ros_workspace
Source space: /home/asd/ros_workspace/src
Build space: /home/asd/ros_workspace/build
Devel space: /home/asd/ros_workspace/devel
Install space: /home/asd/ros_workspace/install
####
#### Running command: "cmake /home/asd/ros_workspace/src -DCATKIN_DEVEL_PREFIX=/home/asd/ros_workspace/devel -DCMAKE_INSTALL_PREFIX=/home/asd/ros_workspace/install -G Unix Makefiles" in "/home/asd/ros_workspace/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/asd/ros_workspace/devel
-- Using CMAKE_PREFIX_PATH: /home/asd/ros_workspace/devel;/opt/ros/kinetic
-- This workspace overlays: /home/asd/ros_workspace/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/asd/ros_workspace/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - lidar_camera_calibration
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'lidar_camera_calibration'
-- ==> add_subdirectory(lidar_camera_calibration)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/asd/ros_workspace/build
####
#### Running command: "make -j8 -l8" in "/home/asd/ros_workspace/build"
####

john-le1996, 2021-03-26:
sensor_msgs/CameraInfo                [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage           [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image                     [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/LaserScan                 [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/PointCloud2               [1158d486dd51d683ce2f1be655c3c181]
             theora_image_transport/Packet         [33ac4e14a7cff32e7e0d65f18bb410f3]
             velodyne_msgs/VelodyneScan            [50804fc9533a0e579e6322c04ae70566]
topics:      /diagnostics                                                  125 msgs    : diagnostic_msgs/DiagnosticArray      
             /image_view/output                                           1895 msgs    : sensor_msgs/Image                    
             /image_view/parameter_descriptions                              1 msg     : dynamic_reconfigure/ConfigDescription
             /image_view/parameter_updates                                   1 msg     : dynamic_reconfigure/Config           
             /rosout                                                      1918 msgs    : rosgraph_msgs/Log                     (8 connections)
             /rosout_agg                                                  1898 msgs    : rosgraph_msgs/Log                    
             /scan                                                         626 msgs    : sensor_msgs/LaserScan                
             /usb_cam/camera_info                                         1896 msgs    : sensor_msgs/CameraInfo               
             /usb_cam/image_raw                                           1895 msgs    : sensor_msgs/Image                    
             /usb_cam/image_raw/compressed                                1896 msgs    : sensor_msgs/CompressedImage          
             /usb_cam/image_raw/compressed/parameter_descriptions            1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/compressed/parameter_updates                 1 msg     : dynamic_reconfigure/Config           
             /usb_cam/image_raw/compressedDepth/parameter_descriptions       1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/compressedDepth/parameter_updates            1 msg     : dynamic_reconfigure/Config           
             /usb_cam/image_raw/theora                                    1901 msgs    : theora_image_transport/Packet        
             /usb_cam/image_raw/theora/parameter_descriptions                1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/theora/parameter_updates                     1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager/bond                                378 msgs    : bond/Status                           (4 connections)
             /velodyne_nodelet_manager_driver/parameter_descriptions         1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_driver/parameter_updates              1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager_laserscan/parameter_descriptions      1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_laserscan/parameter_updates           1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager_transform/parameter_descriptions      1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_transform/parameter_updates           1 msg     : dynamic_reconfigure/Config           
             /velodyne_packets                                             626 msgs    : velodyne_msgs/VelodyneScan           
             /velodyne_points

john-le1996, 2021-03-26:
running roslaunch lidar_camera_calibration play_rosbag.launch bagfile:=lidar_camera.bag

[FATAL] [1616777091.643641127]: Error opening file: /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag
[player-2] process has died [pid 6941, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play --clock --loop /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag __name:=player __log:=/home/asd/.ros/log/95715760-8e52-11eb-8ccb-04ea56f154b9/player-2.log].
log file: /home/asd/.ros/log/95715760-8e52-11eb-8ccb-04ea56f154b9/player-2*.log

heethesh, 2021-03-27:
Is your bag file at this path? `/home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/lidar_camera.bag`, if not move it there.

john-le1996, 2021-03-27:
thanks for the update sir 
but still got this fatal error

NODES
  /
    player (rosbag/play)

ROS_MASTER_URI=http://localhost:11311

process[player-1]: started with pid [12979]
[ INFO] [1616820330.079951953]: Opening /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/new_rectified.bag
[FATAL] [1616820330.080143644]: Error opening file: /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/new_rectified.bag
[player-1] process has died [pid 12979, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play --clock --loop /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles/new_rectified.bag __name:=player __log:=/home/asd/.ros/log/66a4b2a6-8ea7-11eb-a159-04ea56f154b9/player-1.log].
log file: /home/asd/.ros/log/66a4b2a6-8ea7-11eb-a159-04ea56f154b9/player-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

can i contact you sir through email?

john-le1996, 2021-03-27:
thanks for update sir
but still got this fatal error 
can i contact to you through email?

NODES
  /
    player (rosbag/play)

auto-starting new master
process[master]: started with pid [15615]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5387355e-8ebb-11eb-a159-04ea56f154b9
process[rosout-1]: started with pid [15628]
started core service [/rosout]
process[player-2]: started with pid [15632]
[ INFO] [1616822078.195307620]: Opening /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/bagfiles/2016-11-22-14-32-13_test_updated.bag
[FATAL] [1616822078.195542898]: Error opening file: /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/bagfiles/2016-11-22-14-32-13_test_updated.bag
[player-2] process has died [pid 15632, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play --clock --loop /home/asd/ros_workspace/src/lidar_camera_calibration/bagfiles//home/asd/ros_workspace/src/bagfiles/2016-11-22-14-32-13_test_updated.bag __name:=player __log:=/home/asd/.ros/log/5387355e-8ebb-11eb-a159-04ea56f154b9/player-2.log].
log file: /home/asd/.ros/log/5387355e-8ebb-11eb-a159-04ea56f154b9/player-2*.log

john-le1996, 2021-03-27:
thanks for response

closed 2021-03-30 by heethesh

lipeize-pku, 2021-07-28:
I'm having the same issue. I can confirm the bag is there, but I still get "Error opening file".