heethesh/lidar_camera_calibration #33: rosrun lidar_camera_calibration update_camera_info.py <original_file.bag> <calibration_file.yaml>

john-le1996, 2021-03-30:


john-le1996, 2021-03-30:
rosbag info file.bag

types:       bond/Status                           [eacc84bf5d65b6777d4c50f463dfb9c8]
             diagnostic_msgs/DiagnosticArray       [60810da900de1dd6ddd437c3503511da]
             dynamic_reconfigure/Config            [958f16a05573709014982821e6822580]
             dynamic_reconfigure/ConfigDescription [757ce9d44ba8ddd801bb30bc456f946f]
             rosgraph_msgs/Log                     [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/CameraInfo                [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage           [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Image                     [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/LaserScan                 [90c7ef2dc6895d81024acba2ac42f369]
             sensor_msgs/PointCloud2               [1158d486dd51d683ce2f1be655c3c181]
             theora_image_transport/Packet         [33ac4e14a7cff32e7e0d65f18bb410f3]
             velodyne_msgs/VelodyneScan            [50804fc9533a0e579e6322c04ae70566]
topics:      /diagnostics                                                  40 msgs    : diagnostic_msgs/DiagnosticArray      
             /rosout                                                      430 msgs    : rosgraph_msgs/Log                    
             /rosout_agg                                                  414 msgs    : rosgraph_msgs/Log                    
             /scan                                                        196 msgs    : sensor_msgs/LaserScan                
             /usb_cam/camera_info                                         412 msgs    : sensor_msgs/CameraInfo               
             /usb_cam/image_raw                                           412 msgs    : sensor_msgs/Image                    
             /usb_cam/image_raw/compressed                                413 msgs    : sensor_msgs/CompressedImage          
             /usb_cam/image_raw/compressed/parameter_descriptions           1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/compressed/parameter_updates                1 msg     : dynamic_reconfigure/Config           
             /usb_cam/image_raw/compressedDepth/parameter_descriptions      1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/compressedDepth/parameter_updates           1 msg     : dynamic_reconfigure/Config           
             /usb_cam/image_raw/theora                                    417 msgs    : theora_image_transport/Packet        
             /usb_cam/image_raw/theora/parameter_descriptions               1 msg     : dynamic_reconfigure/ConfigDescription
             /usb_cam/image_raw/theora/parameter_updates                    1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager/bond                               120 msgs    : bond/Status                          
             /velodyne_nodelet_manager_driver/parameter_descriptions        1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_driver/parameter_updates             1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager_laserscan/parameter_descriptions     1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_laserscan/parameter_updates          1 msg     : dynamic_reconfigure/Config           
             /velodyne_nodelet_manager_transform/parameter_descriptions     1 msg     : dynamic_reconfigure/ConfigDescription
             /velodyne_nodelet_manager_transform/parameter_updates          1 msg     : dynamic_reconfigure/Config           
             /velodyne_packets                                            196 msgs    : velodyne_msgs/VelodyneScan           
             /velodyne_points                                             196 msgs    : sensor_msgs/PointCloud2

KnightHarute, 2021-10-14:
I faced the same problem, did you slove it?

KnightHarute, 2021-10-14:
I faced the same problem. Did you solve it?