heethesh/lidar_camera_calibration #37: Camera caliberatiion is launching but screen is black in gui

islamtalha01, 2021-09-06:
HI heathenish,
 I am using a USB camera and os1 64 lidars for calibration. I recorded the bag file including the topics that are mentioned in the attachment. Now when I am using the camera calibration lanch file just black windows open and nothing happens. Now I am not able to move forward. Now I am doing an intern in which I have to generate calibrated data set for a deep learning algorithm, where first I require camera and lidar intrinsic and extrinsic parameters.
So Just tell me the steps that I have to follow to get the matrices.
I have checked the issue portion in the GitHub where you solve all the issues of the persons, I want you to tell me the set of steps that I need to follow to get my camera and lidar calibrated. Again mentioning my lidar is ouster os1 64 and camera is a USB camera. so sir please tell me steps you think to approach the end result of calibration keeping the experience of issues of people with ouster in mind. So i don't have to mention individual step issues again and again. I will be very very grateful to you.
From recording to getting rigid transform I need step by step procedure. 
I would be very grateful to you if you conclude the whole procedure in reply. I have been searching and using a lot of open-source ros package but no one gonna work for ouster, your one is supposed to run with ouster. Now I think I find the right one but I need your teaching please help me.
I recorded my ros bag file simply and I don't know about timestamp and how it is set so kindly tell me how to properly record the rose bag file for this. I would be very thankful to you for this favor and I would document all the steps with pics and procedure and will put on your GitHub so others also follow this step by step. I need your help brother. We can also meet on google meet and you can just explain the whole procedure in 10 to 15 min, I will be much thankful to you. I am very thankful to you if you ans this asap as i have to make data set and use the algorithm afterwards probably in the internship era.
 Again thank you very much.

tuoyan1234, 2021-12-13:
I aslo have the problem, do you solve the problem？ and how to solve it? Thx

Chri1996, 2022-10-10:
I've  got the same problem. Any recommendations?

heethesh, 2024-04-10:
Apologies for the inconsistencies, this was developed a long time back and has not been maintained since. I'm working on a new version that uses Open3D point picker and hopefully should resolve these issues. This new version should support more LiDARs and ROS2 as well.