heethesh/lidar_camera_calibration #43: No consistency when using calibrate_camera_lidar.py

jamesheatonrdm, 2021-11-05:
Hi,

I am attempting to use this package and among other issues, there is absolutely no consistency when running calibrate_camera_lidar.py and pressing enter to choose points. Sometimes I start the node, press enter and then I can select points, but then when I close the windows and select a new pose pressing enter does nothing. Sometimes I start the node and pressing enter does not bring up the windows and it's really annoying having to start and stop the node 5 times in the hope that it will work just to pick points for one pose. What could be causing this? How can I fix this?

Rosa712, 2024-04-02:
I have the sam problem, how to fix it?

heethesh, 2024-04-10:
Apologies for the inconsistencies, this was developed a long time back and has not been maintained since. I'm working on a new version that uses Open3D point picker and hopefully should resolve these issues.