heethesh/lidar_camera_calibration #47: Cannot pick points in matplotlib GUI after hit enter key

ajiang17, 2022-04-26:
Hi：
    I run this code , after hit enter key ,I can see point cloud and related rgb image showed up.
    But  I cannot pick  any points in two figures.    left click mouse、right click 、enter key .. all not working .  And also I cannot rotate the pointcloud in 3D view, it's just  a 2D picture . Check following pics for details.
    Can someone help me what is wrong ?
    And I‘m using x11 to forword the  matplotlib GUI  to win10,  is this the reason?  
    Thanks for help! 
  
[image]

jasonCastano, 2022-06-23:
Hey, I have the same problem as you. Did you solved it?

heethesh, 2022-06-23:
Not sure if anything has changed with matplotlib over the years, but I'll probably shift to using Open3D GUI to pick points.

dylantzx, 2022-07-02:
I faced the same issue and downgraded matplotlib to version 3.1.2 and am able to rotate and pick points. I am facing another issue after picking the points though...

ajiang17, 2022-07-08:
En，i give up. Hope someone else can help.

---- 回复的原邮件 ----
| 发件人 | Dylan ***@***.***> |
| 日期 | 2022年07月03日 00:13 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.******@***.***> |
| 主题 | Re: [heethesh/lidar_camera_calibration] Cannot pick points in matplotlib GUI after hit enter key (Issue #47) |

I faced the same issue and downgraded matplotlib to version 3.1.2 and am able to rotate and pick points. I am facing another issue after picking the points though...

ajiang17, 2022-07-08:
> Hey, I have the same problem as you. Did you solved it?

No ，I didn't .Sorry

ajiang17, 2022-07-08:
> I faced the same issue and downgraded matplotlib to version 3.1.2 and am able to rotate and pick points. Hope this helps
@jasonCastano You can try what @dylantzx  did which may help  you.