heethesh/lidar_camera_calibration #52: IndexError: index 1 is out of bounds for axis 1 with size 1

aristow1, 2022-11-15:
im getting this Error after i pick the points on both the 2D and 3D GUI. i tried printing out `error[: `,0]` but `error[: ,1]` is not printing and i dont know why ? ANY HELP PLEASE.

`[INFO] [1668517606.916453]: PCL: (1.4462158679962158, -0.09864430874586105, 0.25353577733039856)
[INFO] [1668517609.022213]: PCL: (1.4393858909606934, 0.4945423901081085, -0.2555065453052521)
[INFO] [1668517610.709231]: PCL: (1.5937378406524658, 0.8632069230079651, 0.1303425431251526)
[INFO] [1668517612.901027]: PCL: (1.6220049858093262, 0.26977214217185974, 0.6466799378395081)
[INFO] [1668517619.628229]: PCL: (1.4517282247543335, -0.09007080644369125, 0.25428515672683716)
[INFO] [1668517623.025287]: IMG: (913.1363636363636, 533.2662337662338)
[INFO] [1668517624.806077]: IMG: (458.5909090909091, 971.5779220779222)
[INFO] [1668517627.325549]: IMG: (254.0454545454545, 653.3961038961039)
[INFO] [1668517629.269590]: IMG: (650.1493506493507, 254.0454545454545)
[INFO] [1668517631.637519]: IMG: (916.3831168831169, 533.2662337662338)
[WARN] [1668517634.823830]: Updating file: /home/kamag/simulation_ws/src/lidar_camera_calibration/calibration_data/lidar_camera_calibration/pcl_corners.npy
[WARN] [1668517636.159035]: Updating file: /home/kamag/simulation_ws/src/lidar_camera_calibration/calibration_data/lidar_camera_calibration/img_corners.npy
[[[-5.15619092  4.32390595]]

 [[-2.13842695  1.23861138]]

 [[ 3.17399677 -5.70847668]]

 [[-6.79263719 -3.41894869]]

 [[ 4.06470721  0.13884155]]

 [[-3.84455849  2.99119131]]

 [[ 4.57203483  5.06013865]]

 [[ 2.34181313 -5.27779634]]]
[[-5.15619092  4.32390595]
 [-2.13842695  1.23861138]
 [ 3.17399677 -5.70847668]
 [-6.79263719 -3.41894869]
 [ 4.06470721  0.13884155]
 [-3.84455849  2.99119131]
 [ 4.57203483  5.06013865]
 [ 2.34181313 -5.27779634]]
[ERROR] [1668517636.201317]: bad callback: <bound method Subscriber.callback of <message_filters.Subscriber object at 0x7f43bf757a10>>
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 76, in callback
    self.signalMessage(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 302, in add
    self.signalMessage(*msgs)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/message_filters/__init__.py", line 58, in signalMessage
    cb(*(msg + args))
  File "/home/kamag/simulation_ws/src/lidar_camera_calibration/scripts/calibrate_camera_lidar.py", line 485, in callback
    calibrate()
  File "/home/kamag/simulation_ws/src/lidar_camera_calibration/scripts/calibrate_camera_lidar.py", line 342, in calibrate
    print(error[:, 1])
IndexError: index 1 is out of bounds for axis 1 with size 1`

unitrixx, 2022-12-08:
Hi aristow,

I just encountered the same problem. As you can see in your console output of the error array, it seems to have the shape (8,1,2) and I think the code assumes it has shape (8,2). So for me writing `error[:,0,0]` and `error[:,0,1]` instead of `error[:,0]` and `error[:,1]` seems to fix the problem.

unitrixx, 2022-12-08:
The actual lines where you might need to do this are:

https://github.com/heethesh/lidar_camera_calibration/blob/688397c500967a42b2aca82d6c11393dd73aaa9c/scripts/calibrate_camera_lidar.py#L340

and

https://github.com/heethesh/lidar_camera_calibration/blob/688397c500967a42b2aca82d6c11393dd73aaa9c/scripts/calibrate_camera_lidar.py#L358

closed 2023-01-24 by aristow1